i2Nav-WHU/KF-GINS #2: please describe dataset logger hardware specs

iceberg1369, 2022-10-22:
I'm so grateful for your work and open sourcing this software

it seems like you have used ICM-20605 IMU in datasets logger. the original IMU raw data are in 1000Hz and trapezoidal integrated to achieve 200 Hz delta data. if this assumption is true. please describe trapezoidal integration algorithm which is carried on raw data to gather datasets

liqiang1026, 2022-10-24:
During ICM20602 raw data collection, we summed every 5 incremental raw data(incremental angle or velocity in origin sampling duration) into one IMU data (data in our datasets), which decreases the data rate from 1000Hz to 200Hz. And the incremental raw data is integrated by the trapezoidal integration algorithm.

iceberg1369, 2022-10-24:
is there any chance hardware logger go open source?

liqiang1026, 2022-10-24:
Sorry to tell you that the hardware data-collecting program will not be open-sourced for the time being.

closed 2022-10-24 by iceberg1369

Dennissy23, 2023-05-15:
> 在ICM20602原始数据收集期间，我们将每5个增量原始数据（原点采样持续时间中的增量角度或速度）汇总为一个IMU数据（数据集中的数据），从而将数据速率从1000Hz降低到200Hz。增量原始数据通过梯形积分算法进行积分。
假如使用梯形积分得到增量形式的数据，那对应时间戳应该选择哪个数据的时间呢，是梯形积分第一个数据的时间还是第二个数据的时间还是两个时间的中间值？

iceberg1369, 2023-05-15:
@Dennissy23 do you know howicm-20602 registers is configured?

wang-lq, 2023-05-16:
> > 在ICM20602原始数据收集期间，我们将每5个增量原始数据（原点采样持续时间中的增量角度或速度）汇总为一个IMU数据（数据集中的数据），从而将数据速率从1000Hz降低到200Hz。增量原始数据通过梯形积分算法进行积分。
> > 假如使用梯形积分得到增量形式的数据，那对应时间戳应该选择哪个数据的时间呢，是梯形积分第一个数据的时间还是第二个数据的时间还是两个时间的中间值？

用5个速率型的数据积分得到1个增量形式的数据，这个增量型数据的时间应该给第5个数据的时间

Wanghang448, 2024-11-19:
> 很遗憾地告诉您，硬件数据收集程序暂时不会开源。

您好，这个开源代码只支持200Hz的IMU数据吗

wang-lq, 2024-11-19:
理论上可以支持任意频率的IMU数据

Wanghang448, 2024-11-19:
> 理论上可以支持任意频率的IMU数据

请问我的输出结果只有前面3秒，后面全部是nan，是什么原因