i2Nav-WHU/KF-GINS #5: Accept input data from low-cost MEMS IMU

shockjiang, 2023-01-19:
A lot of users afford low-cost MEMS IMU in their project or product. It would be of great helpful if you provide the way to accept input data from low-cost MEMS imu, which uses linear acc and angular velocity directly instead of their delta.

What's more, I wonder whether the noise parameter of IMU should be adapted if we use linear acc and angular velocity as input.

shockjiang, 2023-01-19:
I compared a MEMS IMU with your IMU in dataset/kf-gins.yaml, which seems that MEMS IMU is much better than yours. very confusing. What's wrong here?

- MEMS IMU:  wheeltec n100 (~400rmb), https://item.jd.com/10044417308654.html
  - as we can get the noise from the data spec, after converting:
  -  imunoise:
 ```
  arw: [0.168, 0.168, 0.168]          # [deg/sqrt(hr)], 0.0028deg/s/sqrt(hz) in spec
  vrw: [0.0441, 0.0441, 0.0441]          # [m/s/sqrt(hr)], 75ug/sqrt(hz) in spec
  gbstd: [5., 5., 5.]        # [deg/hr]
  abstd: [400.0, 400.0, 400.0]     # [mGal]
  gsstd: [100.0, 100.0, 100.0]  # [ppm]
  asstd: [100.0, 100.0, 100.0]  # [ppm]
```

- IMU you used in your testing dataset:
imunoise:
```
  arw: [0.24, 0.24, 0.24]          # [deg/sqrt(hr)]
  vrw: [0.24, 0.24, 0.24]          # [m/s/sqrt(hr)]
  gbstd: [50.0, 50.0, 50.0]        # [deg/hr]
  abstd: [250.0, 250.0, 250.0]     # [mGal]
  gsstd: [1000.0, 1000.0, 1000.0]  # [ppm]
  asstd: [1000.0, 1000.0, 1000.0]  # [ppm]
```

wang-lq, 2023-02-05:
The MEMS IMU you used is indeed better than ours. This MEMS IMU in our testing dataset is ICM20602 of InvenSense, of which the retail price is about 30~50 RMB. Besides, The provided imu noise parameters are optimized by our experiments,  which are not the same as that in the datasheet.

closed 2023-02-08 by wang-lq