i2Nav-WHU/KF-GINS #12: How to log our own dataset ?

mayankpatel97, 2023-08-17:
[image]
Hi, I have a setup of IMU and GNSS with RTK, which logs acceleration in m/sec2 and rate of change in degree/second. 
In the data set the given IMU format requires XYZ incremental angles in radian and XYZ incremental velocity in m/s.
 Please guide me, how can I get these values from accelerometer and Gyroscope.

mayankpatel97, 2023-08-22:
[image]
I am getting this issue in nav_result file. 
what could be the possible issue? 

Input files can be downloaded from below link 

https://drive.google.com/drive/folders/1DpdgdxCNFUjG_pic4iYpn3YNzv1MbeXz?usp=sharing

wang-lq, 2023-09-06:
You can multiply your original measurement(m/s^2 for acceleration and rad/s for angular velocity) with the sample duration to get the incremental IMU data.

closed 2023-09-06 by wang-lq

Wanghang448, 2024-11-19:
> ![image](https://private-user-images.githubusercontent.com/92679540/262247978-a7744dbf-ef5b-49a7-9fd9-ed504a1b9a63.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIwMTAwNTcsIm5iZiI6MTczMjAwOTc1NywicGF0aCI6Ii85MjY3OTU0MC8yNjIyNDc5NzgtYTc3NDRkYmYtZWY1Yi00OWE3LTlmZDktZWQ1MDRhMWI5YTYzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTE5VDA5NDkxN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTIzNzZjMjdjYzVkNTM4NGI4ODJjZThmMTAyNzY2ZGRmNmRiMjBhMWEyNTFhOGEyMGQ1YzdkNGZhOTNjMzA5YTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.LwINghnSiSWR1aj8eEYq8sBLLEsWcKCFGOIMCAfEzr8)我在 nav_result 文件中遇到了这个问题。可能的问题是什么？
> 
> 输入文件可以从以下链接下载
> 
> https://drive.google.com/drive/folders/1DpdgdxCNFUjG_pic4iYpn3YNzv1MbeXz?usp=sharing

您好，我也遇到了同样的问题，有大量的数据输出是nan，请问您解决了吗

mayankpatel97, 2024-11-19:
Hi, I have solved the issue. It was an issue with input data. You have to check input data properly. One surged value in the input and the algorithm will go crazy. Log input data check values should be in range make sure the dot in float value should not miss in the input data. It was happening in my case once in 10 minutes and the float value becomes integer and that triggers it in to an unknown state