i2Nav-WHU/KF-GINS #13: How do I calculate imu noise parameters for my IMU

mayankpatel97, 2023-09-13:
Hello,
How do I calculate below parameters for my IMU?
These parameters are given in kf-gins.yaml

imunoise:
  arw: [0.003, 0.003, 0.003]          # [deg/sqrt(hr)]
  vrw: [0.03, 0.03, 0.03]          # [m/s/sqrt(hr)]
  gbstd: [0.027, 0.027, 0.027]        # [deg/hr]
  abstd: [15.0, 15.0, 15.0]     # [mGal]
  gsstd: [300.0, 300.0, 300.0]  # [ppm]
  asstd: [300.0, 300.0, 300.0]  # [ppm]
  corrtime: 4.0                    # [hr]

iceberg1369, 2023-09-13:
@mayankpatel97 you could extract some paramters from datasheet and for ARW and VRW you could do some random walk analysis on the sensor raw data

mayankpatel97, 2023-09-14:
@iceberg1369  Hi, 
Thanks for the response . I would try the suggested solution at my end . 
Should I go for Allan Variance noise analysis?

iceberg1369, 2023-09-14:
@mayankpatel97 yes, allan variance is the correct way, I think a few scripts exists in github. let us know if you find anything. I think it is a good addition to this repo

mayankpatel97, 2023-09-14:
@iceberg1369 
Okay ... Thank You

closed 2024-02-29 by mayankpatel97