i2Nav-WHU/KF-GINS #16: 关于机械编排中的姿态更新

HitComBoy, 2024-01-10:
作者好！想问下机械编排姿态更新部分，利用等效旋转矢量计算中间时刻的四元数的过程中，temp的计算式中是否该加一个q_nk,nk-1呢，也就是我觉得qne_cur.inverse() * qne_pre似乎得到的不是q_ek,ek-1
midvel = (pvapre.vel + pvacur.vel) / 2;
qne_pre   = Earth::qne(pvapre.pos);
qne_cur   = Earth::qne(pvacur.pos);
temp1     = Rotation::quaternion2vector(qne_cur.inverse() * qne_pre); <---------------------------这个地方
qne_mid   = qne_pre * Rotation::rotvec2quaternion(temp1 / 2).inverse();
midpos[2] = (pvacur.pos[2] + pvapre.pos[2]) / 2;
midpos    = Earth::blh(qne_mid, midpos[2]);

wang-lq, 2024-02-22:
你好，代码里面qne具体为q_n^e，e系不对载体位置变化，n系是变化的，所以这里temp1得到的是q_nk-1^nk。

closed 2024-04-25 by wang-lq

ChinaTonyDiao, 2024-06-06:
你好，如果我没有推算错的话，应该是qn(k)^n(k-1) = qe(k)^n(k) * qe(k-1)^e(k) * qn(k-1)^e(k-1)
这里是相对e系来说的吗？ 如果是，qe(k-1)^e(k)为1可忽略；
但是在位置更新中，是相对i系来说的，qe(k-1)^e(k)没有忽略；
所以为什么这里忽略了呢？

wang-lq, 2024-07-05:
原来的解释不考虑e系的变化确实欠妥当。在机械编排更新的公式中，qe(k-1)^e(k)确实不应该忽略，所以位置更新中考虑了qe(k-1)^e(k)。上面提到的那个公式推导里面忽略了qe(k-1)^e(k)，是因为这里本身就是近似计算中间时刻的位置，而且这个近似计算的影响较小，所以就忽略了。