i2Nav-WHU/KF-GINS #17: Add support to ue velocity as input values

iceberg1369, 2024-02-29:
Sice velocity is more accurate measurment in gnss devices. I think this feature will lead to accurate attitude in navigation result despite having positions with high errors.

wang-lq, 2024-04-25:
We will consider it in the following update.

closed 2024-04-25 by wang-lq

iceberg1369, 2024-07-20:
I would like to know is there any update planned for next release?

wang-lq, 2024-07-28:
There will be no new version released recently to add velocity updates. You can refer to some literature on implementing the velocity update algorithm.

Page 97 on Estimation technology of low-cost inertial navigation [chrome-extension://cdonnmffkdaoajfknoeeecmchibpmkmg/assets/pdf/web/viewer.html?file=https%3A%2F%2Fwww.ucalgary.ca%2Fengo_webdocs%2FNES%2F05.20219.EHShin.pdf]