i2Nav-WHU/KF-GINS #18: How to Identify Yaw Angle Drift? What are the Key Parameters to Monitor in the Code ?

mayankpatel97, 2024-02-29:
Thank you for open-sourcing this excellent code.

I've implemented modifications to KF-GINS to ensure compatibility with the STM32 Microcontroller at my end. The code now effectively processes sensor inputs and executes the Kalman process in real-time, functioning seamlessly.

I do have a specific inquiry. When the device remains in a static state for several minutes, there is noticeable drift in the Yaw angle, which is expected given its reliance on a 6DoF system. And, EKF corrects this drift when the device is in motion.

To identify whether the yaw angle has drifted, which parameters should I monitor in the code?

iceberg1369, 2024-03-01:
Could you please share your stm32 implemented project?

YoungCapta1n, 2024-03-26:
> @mayankpatel97 the only way you have to invalidate yaw angle in stillness mode, you have to consider yaw angle inaccurate after 120 seconds . The other way is to implement zero-velocity algorithm. By the wayCould you please share your stm32 implemented project?

Thank you for providing this excellent code about GINS. I have performed some movement test using the KF-GINS, based on dual-antenna GNSS and MEMS IMU. It can be seen from the following figures that some difference between GNSS heading and GINS heading. I was wondering that is this normal in the GNSS inertial results?

[image]
[image]

iceberg1369, 2024-03-27:
@YoungCapta1n what gnss oem you used?

YoungCapta1n, 2024-03-27:
> @YoungCapta1n what gnss oem you used?

Unistrong V28U

iceberg1369, 2024-03-27:
@YoungCapta1n is your dataset IMU and GNSS data synchronized?

YoungCapta1n, 2024-03-27:
> @YoungCapta1n is your dataset IMU and GNSS data synchronized?
I have reimplemented the code using C-99, and the KF-GINS loop have been shown in the following screen shot. I think the IMU and GNSS data are time synchronized. Maybe there is some errors in my code.

[image]

wang-lq, 2024-04-25:
For the single GNSS antenna configuration, the yaw angle is unobservable when the carrier is stationary or moving at a constant speed. Under these conditions, you can determine that the yaw angle will drift. However, for the double antenna configuration, GNSS can output yaw measurements and keep the yaw angle observable.

closed 2024-04-25 by mayankpatel97