i2Nav-WHU/KF-GINS #26: 如何理解融合后结果中的roll-pitch-yaw呢？

lyk125, 2024-07-22:
hello:
    非常感谢开源的工程和课程！
     使用dataset里的数据运行后，发现融合后结果的yaw角度很奇怪，是在0~360之间循环，truth.nav里的yaw也是这样：
[image]
      这个现象怎么理解呢？融合后结果的attitude angles (roll-pitch-yaw)的坐标系是哪个呢？
    期待您的答复!

regards

wang-lq, 2024-08-09:
融合后的姿态角表示n系和b系之间的相对旋转，具体是表示n系分别按照Z轴、Y轴、X轴旋转航向角、俯仰角和横滚角后得到的坐标姿态（也就是b系）。欧拉角的具体定义请参考我们组开源姿态算法讲义。

closed 2024-08-09 by wang-lq

lyk125, 2024-08-15:
谢谢答复！但是我还是没有理解为什么yaw航向角一直在递增？麻烦指点下

wang-lq, 2024-08-15:
你是说你用KF-GINS跑出来的航向一直在递增嘛？超过了0~360的范围？

lyk125, 2024-08-19:
是的，我看到真值truth.nav里的yaw角也是在一直递增，在0~360之间循环
[image]

wang-lq, 2024-08-20:
yaw递增不是很正常吗，载体的航向在变化，航向角度增加的减小是正常的。你可以再观察一下，肯定还是有航向递减的现象的。发自我的 iPhone

image.png (view on web)

lyk125, 2024-08-22:
哦哦，明白了：是b系一直在运动，运动中方向在变，yaw在变换