i2Nav-WHU/KF-GINS #29: 近似计算更新Q矩阵疑问

ChinaTonyDiao, 2024-12-31:
最近在拜读误差反馈卡尔曼滤波算法设计的时候，发现一个疑问
在讲义中Q矩阵通过离散后表达式为
[image]
对应ppt中
[image]
但是在代码实现中我发现
[image]

这里
[image]
这一部分使用了上一时刻的值近似，
我的猜想是：q矩阵不随时间变化，忽略；但对于G矩阵，因为目前还在预测阶段，我们不知道k时刻确切的方向余弦Cnb去更新G矩阵，所以使用上一时刻的G继续。

wang-lq, 2025-01-08:
可以用最新时刻的Cnb更新G阵，但是因为IMU采样频率足够高，可以认为两个时刻的G阵差别很小，这在工程上是一个可以接受的近似。

ChinaTonyDiao, 2025-01-10:
好的谢谢😊

closed 2025-01-10 by ChinaTonyDiao